VasilKalchev/LiquidMenu #1: Can't compile on ESP8266 boards

sindudas, 2016-08-07:
.../Documents/Arduino/libraries/LiquidMenu/src/LiquidLine.cpp: In member function 'bool LiquidLine::attach_function(uint8_t, void (*)())':
.../Documents/Arduino/libraries/LiquidMenu/src/LiquidLine.cpp:37:21: error: cast from 'LiquidLine*' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
  print_me((uint16_t)this);
                     ^
.../Documents/Arduino/libraries/LiquidMenu/src/LiquidLine.cpp: In member function 'bool LiquidLine::set_focusPosition(Position, uint8_t, uint8_t)':
.../Documents/Arduino/libraries/LiquidMenu/src/LiquidLine.cpp:51:21: error: cast from 'LiquidLine*' to 'uint16_t {aka short unsigned int}' loses precision [-fpermissive]
  print_me((uint16_t)this);

VasilKalchev, 2016-08-16:
It should be fixed now.

sindudas, 2016-08-16:
Now it don't compile again but different error message:

xtensa-lx106-elf-gcc: error: libraries\LiquidMenu\LiquidMenu.a: No such file or directory
exit status 1

TyIsI, 2018-03-07:
@sindudas your error is due to lack of support of the dot_a_linkage flag in library.properties by the ESP8266/ESP32 builder/compiler. Changing dot_a_linkage to false in your local copy of library.properties will likely fix your problem.

@VaSe7u FYI

closed 2018-03-08 by VaSe7u