VasilKalchev/LiquidMenu #2: LiquidMenu RAM memory problem

Geusantropy, 2016-08-19:
hello, due to some problems with a too low amount of RAM I've tried to use the Flash memory to store the text lines present inside the menu.. but I had some problems.  I've tried to use the PROGMEM command and the relative methods to recall stored data, but the printet characters on lcd seems to be wrong converted (numbers or simbols). Can you explain the correct way to put flash stored text inside a liquidline object? Please, be detailed... I'm just a mechanical engineer :-)

VasilKalchev, 2016-08-22:
Hi, I added support for printing text lines stored in the flash memory as `const char[]`.
The way to do it is by defining the text line like this:
```c++
const char text[] PROGMEM = "Some text";
```
then attach it to the `LiquidLine` object as usual:
```c++
LiquidLine someLine(0, 0, text);
```
and finally call the function:
```c++
someLine.set_asProgmem(1);
```
this function tells the object that this variable is stored in the flash. '1' is the consecutive number of the variable for this `LiquidLine` object.

This update is released as beta [here](https://github.com/VasilKalchev/LiquidMenu/releases/tag/1.1.6-beta).

closed 2016-08-30 by VasilKalchev

Geusantropy, 2016-08-31:
thanks, works fine