VasilKalchev/LiquidMenu #4: Variable string does not display correctly when used as LiquidLine parameter

joshfeliciano, 2016-12-11:
See title

VasilKalchev, 2016-12-13:
I will be able to test the problem in 5 days.

VasilKalchev, 2017-01-21:
Sorry for the huge delay.

Use `char*` for a changing string:
```
char* varString= "State 1";
LiquidLine line1(0, 0, varString);
...
varString = "State 2";
```
You can cast the string literal to `(char*)` to avoid the warning.

closed 2017-02-07 by VasilKalchev

joshfeliciano, 2017-02-08:
Thanks much!

Fluxia0, 2018-11-26:
hello 
Arduino's compiler seem to complain when doing this : 
"_deprecated conversion from string constant to 'char*' [-Wwrite-strings] char* varString= "State 1";"_
Also I have tried using a pointer to const chars but the strings do not display correctly in the LCD.

VaSe7u, 2018-11-30:
Casting the string literal to `(char*)` (e.g. `char* varString = (char*)"State 1";`) should hide the warning.
A more correct way would be to assign the pointer to a char array in memory.