VasilKalchev/LiquidMenu #12: Wrong cast

nfhktwrbq, 2018-03-08:
In class liquidLine incorrect cast for char pointer:
`char* variable = *reinterpret_cast<char**>( const_cast<void*>(_variable[number]) );`

correct variant is:
`char* variable = const_cast<char*>(reinterpret_cast<const char *>(_variable[number]));`

VaSe7u, 2018-03-09:
Thank you, @nfhktwrbq.

nlurkin, 2020-12-23:
Hi, 
I am facing the same issue here. If i use the second variant it works correctly. I see that you implemented it and then reverted a few months later. Any reason why the "correct variant" was not correct?

VaSe7u, 2021-11-02:
Hello nlurkin,

I reverted it because the examples were using a different way of handling dynamic text. But this way really is better and is now re-reverted.

PR #53 (2021-11-02): Simplified dynamic text variables.
Usage of dynamic text as a variable for `LiquidLine` object is simplified.
Resolves #12 and #41.

Character array variables inside `LiquidLine` objects are now correctly "casted" back for printing. Previous usage required creating and passing a `char*` to `LiquidLine` and pointing that variable to the actual text. Current usage allows passing a character array directly as a variable to `LiquidLine` objects and changing its value directly (using `strncpy`).
closed 2021-11-02 by VaSe7u