VasilKalchev/LiquidMenu #25: switch_focus return bool

Fettkeewl, 2019-04-03:
**Is your feature request related to a problem? Please describe.**
I am trying to implement menu control using only 1 rotary encoder with push-button built-in.
I'm struggling with making a menu where there are some MenuLines that do not have a function attached but merely act as a "title" and have the variable below it.
Every MenuScreen is 2 MenuLines only, 

- MenuLine1 is a title

- MenuLine 2 is a variable to change.

Currently encoders left/right swaps screens, When the focus switches via encoder button press left/right becomes functioncallers untill i push the button to end the focus. 

The issue in short as I interpret it:
After some presses on the button, the "title"/MenuLine1 gets focused and has no function and I get blocked from returning to the left/right navigation function.

**Describe the solution you'd like**
Change: 
**void** LiquidMenu::switch_focus | ( | bool | forward = true | ) |  

to:
**bool** LiquidMenu::switch_focus | ( | bool | forward = true | ) |  

Make the return true if the previous/next MenuLine focused has 1 or more functions attached, false otherwise.

**Describe alternatives you've considered**
alternatively skip "switch_focus" of MenuLines that do not have functions attached, there is nothing to focus then?

**Additional context**
I have not played enough with your library to fully understand it, I wonder if I am taking a wrong approach here, perhaps it's not ment to have "dumb title MenuLines" combined with "function attached MenuLines". If above is not possible, I'd love any suggestion to control LiquidMenu with only 1 rotary encoder.

*Remember to use the labels.*

**Current state of my loop**
```
void loop() {
  // Interrupt for encoder
  if (intrSwitch || intrEncod)
  {
    if (intrSwitch) 
    {
      encoderResult=readEncoderButton();
      switch(encoderResult)
      {
        case 1:
          Serial.println("Menu press");
          menu.switch_focus();
          isFocused = !isFocused;
          break;
        case 2:         
          Serial.println("Menu hold");
          //menu.call_function(1);
          break;
      }
      intrSwitch = 0;
    }
    else if (intrEncod)
    {
      encoderResult=readEncoder();
      switch(encoderResult)
      {
        case 1:
          Serial.println("Menu Previous");
          if (isFocused)
            menu.call_function(1);
          else          
            menu--;
          break;
        case 2:
          Serial.println("Menu Next");
          if (isFocused)
            menu.call_function(2);
          else
            menu++;
          break;     
      } 
      intrEncod = 0;
    } 
    lastActionMillis = millis();    
  }
}
```
**End result I wish for I guess.**
**Changing this**
```
case 1:
          Serial.println("Menu press");
          menu.switch_focus();
          isFocused = !isFocused;
          break;
```
**To this**
```
case 1:
          Serial.println("Menu press");          
          isFocused = menu.switch_focus();
          break;
```

VaSe7u, 2019-04-03:
Hi, @Fettkeewl.
It is perfectly fine to have "dumb title MenuLines" combined with "function attached MenuLines" on one screen.
You can check if the 'line' is "function attached" or "title" line by checking the return of the function `bool LiquidMenu::call_function(uint8_t number)`. It returns `true` if it called the attached function (i.e. "function attached line" and `false` if there wasn't an attached function to call (i.e. "title line"):

```c++
switch (rotaryInputType) {
	case rotaryButton: { // switch focus
		// Button press cycles the focus.
		menu.switch_focus();
	} break;
	case rotaryLeft: { // left screen or 'left' function
		bool callable = menu.call_function(1); // try to call 'left' function
		if (callable == true) {
			// The function was called successfully, so this is a
			// 'callable line'. Nothing else to do.
		} else {
			// No function to call, so this is is a 'title line',
			// here rotation left means previous screen.
			menu.previous_screen();
		}
	} break;
	case rotaryRight: { // right screen or 'right' function
		bool callable = menu.call_function(2); // try to call 'right' function
		if (callable == true) {
			// The function was called successfully, so this is a
			// 'callable line'. Nothing else to do.
		} else {
			// No function to call, so this is a 'title line',
			// here rotation right means next screen.
			menu.next_screen();
		}
	} break;
	default: {
		// Invalid rotaryInputType.
	} break;
}
```

There is a potential problem with this method, if for some reason there is a 'line' with only one attached function. In that case you can attach a dummy function, to prevent changing screens when on a 'function attached line'.

The solution you proposed is more intuitive, I'll look into it for a future update.

Fettkeewl, 2019-04-04:
Thank you for your answer and proposal!
I think I can make that work, will try it today.

closed 2019-04-06 by VaSe7u