VasilKalchev/LiquidMenu #27: Snprintf prints jibberish

Fettkeewl, 2019-04-06:
**Describe the bug**
Part of me hopes this is a bug and not a lack of my own coding skills :P!
Im trying to preformat a char array so I can have it look the way I want before
printing it to the screen.

Using snprintf I can format my output and display it in Serial. I have used this way of formatting char strings before the use of liquidmenu to print text the way I want it on an LCD. With liquidmenu I get trash characters displayed.

**To Reproduce**
Try code below

**Screenshots**
[image]
[image]

**Related code**
```
char txtbuffer[17] = "Empty";
LiquidLine test(0, 0, txtbuffer);
LiquidScreen testS(test);

void setup() {
  Serial.begin(9600);
  setupEncoder();
  setupMenu();
  
  Serial.println(txtbuffer);
  snprintf(txtbuffer, 17, "BUFF: %s", "test");
  Serial.println(txtbuffer);
  menu.add_screen(testS);
  menu.update();
  
}

```

**Version of the library**
1.4.0

VaSe7u, 2019-04-06:
What is the output of `txtbuffer` on the display before formatting it with `snprintf(...)`? The `LiquidLine` object uses pointers to char arrays instead of char arrays directly (even though they are similar), take a look at the examples `C_functions_menu` and `D_buttons_menu`.

I think that the `size` argument of `snprintf(...)` should be 16.

Fettkeewl, 2019-04-06:
**First:** according to this http://www.cplusplus.com/reference/cstdio/snprintf/
size should be 17 because the output string is 17-1 due to null terminating character

I tried this below, before snprintf the screen is just blank, after snprintf I get jibberish.
All serial.prints perform normally

```cpp
char txtbuffer[20] = "Empty";
LiquidLine test(2, 0, txtbuffer);
LiquidScreen testS(test);

void setup() {
  Serial.begin(9600);
  setupEncoder();
  setupMenu();
  menu.add_screen(testS);
  menu = testS;  
  Serial.println(txtbuffer);
  menu.update();
  delay(5000);
  snprintf(txtbuffer, 17, "BUFF: %s", "test");
  Serial.println(txtbuffer);
  //menu.add_screen(testS);
  menu.update();
  delay(10000);
}
```

however I tried this aswell and it atleast did output the first Char.
"E" then "B" after snprintf
```cpp
char txtbuffer[20] = "Empty";
LiquidLine test(2, 0, txtbuffer[0]);  //<--------- difference
LiquidScreen testS(test);

void setup() {
  Serial.begin(9600);
  setupEncoder();
  setupMenu();
  menu.add_screen(testS);
  menu = testS;  
  Serial.println(txtbuffer);
  menu.update();
  delay(5000);
  snprintf(txtbuffer, 17, "BUFF: %s", "test");
  Serial.println(txtbuffer);
  //menu.add_screen(testS);
  menu.update();
  delay(10000);
}

VaSe7u, 2019-04-06:
> "I tried this below, before snprintf the screen is just blank, after snprintf I get jibberish."

`char txtbuffer[20]` is set to `"Empty"`. If the screen doesn't print "Empty" (i.e. displays nothing or jibberish), this means that `txtbuffer` isn't working before `snprintf(...)`. This is expected, because the `LiquidLine` object uses pointers to char arrays, not char arrays directly, it doesn't work the same way as `Serial.print(...)` or `lcd.print(...)`.
You should have a separate char pointer that points to `txtbuffer` and use that pointer as an argument to `LiquidLine`'s constructor (examples: `C_functions_menu` and `D_buttons_menu`).

Fettkeewl, 2019-04-07:
Thank you, your suggestion worked great. Although I do not understand it.
textbuffer array is a pointer to the address of the first value of the array in memory, assigning this address to a pointer makes it usable, but using the address directly does not? How weird..

VaSe7u, 2019-04-07:
It is weird, the library is a little hacky. It manually recognizes the data type and stores the address of the variable in a void pointer, then when printing - the variable is casted back. Take a look at `bool LiquidLine::add_variable(T &variable)`, `void LiquidLine::print(DisplayClass *p_liquidCrystal, bool isFocused)` and `recognizeType.cpp` if you want to know how it works.

Basically, for dynamic text, you can use a `char*` that can be pointed to different char arrays. And for constant text, `const char*` directly.

closed 2019-04-07 by VaSe7u