VasilKalchev/LiquidMenu #29: change_screen with LiquidScreen pointer

Cypher315, 2019-04-24:
Hi !

I have a global variable 
`LiquidScreen* previous_screen;`

Before changing screen, I save the current one in this variable (I have a back button on my board which I would like to use to go to the previous screen) I proceed like this :
`previous_screen = menu.get_currentScreen();`

When I press the back button this is called:
`menu.change_screen(previous_screen);`

but when I try to compile I have this error:
`invalid conversion from 'LiquidScreen*' to 'uint8_t {aka unsigned char}' [-fpermissive]`

I don't understand why it tries to call the function with an `uint8_t` as parameter and not this one
 `
  bool change_screen(LiquidScreen &p_liquidScreen);`

I can't use the `previous_screen()` function because my menu is quite long and the screens are not really ordered by number. Sorry if this is a newb question ! 

Have a great day,

Anthony

VaSe7u, 2019-09-22:
Hi, sorry for the delay.

The example `K_getters_menu.ino` seems incorrect. I will be able to test it in a few days and I will get back to you.

In the meantime, have you tried dereferencing the `previous_screen` object when calling the function:
`menu.change_screen(&previous_screen);`

VaSe7u, 2019-12-07:
This was a bug with the library. The method `LiquidScreen* LiquidMenu::get_currentScreen()` was returning a pointer but the method `bool LiquidMenu::change_screen(LiquidScreen &ls)` was expecting a reference.

Now `bool LiquidMenu::change_screen(LiquidScreen *ls)` expects a pointer.

Sorry for the confusion.

closed 2019-12-07 by VaSe7u