VasilKalchev/LiquidMenu #32: LiquidMenu objects have to be defined globally

KevinVanthuyne, 2019-10-14:
I'm using LiquidMenu for a project with the ESP32 with the LiquidCrystal_I2C library. I'm trying to write solid code with separated functionality over different files. For that i want to define LiquidLine and LiquidMenu objects in the constructor of separate classes. When i do this the program crashes as soon as i try to do an operation on the LiquidMenu object. 

Am i overlooking something or is it just not possible with this library?

**Code example**
This works:
```c
LiquidCrystal_I2C lcd(0x27, 20, 4);
LiquidLine exampleLine(0, 0, "Example");
LiquidScreen exampleScreen(exampleLine);

LiquidMenu menu(lcd);

void setup() {
  lcd.init();
  menu.add_screen(exampleScreen);
  menu.update();
}
```

But this doesn't:
```c
void setup() {
  LiquidCrystal_I2C lcd(0x27, 20, 4);
  LiquidLine exampleLine(0, 0, "Example");
  LiquidScreen exampleScreen(exampleLine);
  LiquidMenu menu(lcd);

  lcd.init();
  menu.add_screen(exampleScreen);
  menu.update();
}
```

**Version of the library**
v1.5.0

VaSe7u, 2019-10-14:
Can you be more specific about the crash that you are getting? Does the Arduino hang when you reach `menu.update()` or does it restart. Does it crash when you are trying to call methods of the `LiquidMenu` object in the `setup()` or in the `loop()` function? If you are using the Arduino IDE you can test where the problem occurs by turning on an LED and placing a delay after that. Are you getting compilation warnings or errors?

Are you wrapping the constructors `LiquidLine` and `LiquidMenu` in your own separate constructors or are you wrapping the classes in your own classes? Can you give me a closer example to what you are trying to do, since the only problem I can think of, with the above example is that the object gets destroyed when you leave the function `setup()`.

KevinVanthuyne, 2019-10-15:
I searched some more and it seems that the problem has to do with the ESP32 not being fully compatible with c++ code. Smart pointers can't be used which was what i'm using. So my problem doesn't have to do with the library but with the ESP-IDF, sorry for posting an issue!

closed 2019-10-15 by KevinVanthuyne

VaSe7u, 2019-10-15:
No problem, good luck.