VasilKalchev/LiquidMenu #33: Screen Lines could link to a new Menu

DanielW0830, 2019-11-09:
Was wondering if this sounded like a good request.

When creating a LiquidLine, currently we can attach a function. (that's fine)

But if I want to select a line on a screen which also takes me to a new menu, 
wouldn't it be easier to attach a menu to a liquid line.  (which also makes it selectable)

LiquidLine line( 0, 0, "Config", ConfigMenu);     <----  this also makes it selectable

Then, no callback would be needed.
I would just call call_function(0) to make current menu be the menu passed into the line.

Dan

VaSe7u, 2019-11-28:
If I understand right you are looking for the `LiquidSystem` class. Take a look at this example: [H_system_menu](https://github.com/VaSe7u/LiquidMenu/tree/master/examples/H_system_menu).

Let me know if I misunderstood you.

DanielW0830, 2019-11-29:
Yes, that's a very good example.
I'm suggesting one step further.
In the example you say:   "This method can be wrapped inside a callback function and called on through the menu."

This is good.   But I have an idea.  :)
Let's say Screens has a method to add a menu (or maybe a constructor)
Now if someone selects the screen, instead of calling a callback, the screen would just go to the menu it has attached.  Then when a "back" method is called on a menu, it would go back to the parent menu that it came from.

In this way, navigating multiple menus would be so easy.  
(Btw, I implemented the menu system for some military helicopters.   :)  )

No callbacks would need to be added, and the menus would be smart enough to navigate with just two methods.   :)

What do you think?

Dan

DanielW0830, 2019-11-29:
Another thing I've done in the past is have a line item have a on_changed method.
void  on_changed(LiquidLine *pLiquidLine)
Then, a line could have an inc() & dec() function. 
as it changes, the on_changed gets called an the user can decide what to do when a variable hits a limit.

DanielW0830, 2019-11-29:
In your example, you call this a bunch of times.  All these calls and callbacks could be eliminated.
outputs_line.attach_function(1, goto_outputs_menu);

instead,  switch_focus() would jump right to the attached menu.
and a call to .menu_back() on a screen would get you to the previous menu.
(or the same menu if there is no parent menu attached)

Here's an overview:

menu1
        screen1
	screen2
	screen3
	screen4 ->  (has attached menu,  selecting calls change_focus() ) 
                                                    menu2 
							screen1
							screen2
							screen3  (calling   .menu_back()   goes to menu1,screen4)
	screen5
	screen6

menu3
	screen1
	screen2
	screen3 ->   menu4
			        screen1
			        screen2
			        screen3
			        screen4  (calling   .menu_back()   goes to menu3,screen3)

	screen4 ->   menu5
			         screen1
			         screen2
			         screen3
			         screen4  (calling   .menu_back()   goes to menu3,screen4)

VaSe7u, 2019-12-04:
I agree, this will make it more intuitive to navigate menus. The problem is that this library is using a lot of RAM because it uses static memory for the arrays that hold the functions callbacks, the variables pointers and their types, the `LiquidLine` pointers inside the `LiquidScreen` and so on. To attach a `LiquidMenu` to a `LiquidLine` I'll have to add two more `LiquidMenu` pointers inside `LiquidLine` which will increase the size of every `LiquidLine` object and the statically allocated memory for the callback functions will be less utilized.

I can't add this functionality for now, but I'll definitely add it in the future if I update the library to use dynamic memory.

closed 2019-12-07 by VaSe7u