VasilKalchev/LiquidMenu #34: Switching screens should reset focus item back to first line

DanielW0830, 2019-11-20:
I noticed the menu seems to remember is an item is in focus.
when switching to a new screen, the option to have it reset back to first line would be good.

I'd do it in code, but I have no method which tells me second line is in focus.

VaSe7u, 2019-11-28:
You are right, I'll add accessors for the focus variable.

VaSe7u, 2019-12-01:
These are the new methods:
```c++ 
/// Directly select focused line.
/**
@param lineIndex - index of the focused line
@return true: on success, false: when the selected line doesn't exist or it isn't focusable
*/
bool set_focusedLine(uint8_t lineIndex);

/// Get the index of the currently focused line.
/**
@returns the index of the currently focused line
*/
uint8_t get_focusedLine() const;
```
They apply to classes: `LiquidScreen`, LiquidMenu` and `LiquidSystem`.

The update is not yet released, download it from the master branch.

closed 2019-12-01 by VaSe7u