VasilKalchev/LiquidMenu #36: Error with example for i2c

andrwgldmn, 2020-01-11:
IDE 1.8.10
sketch example for I2C gives this error:
`no matching function for call to 'LiquidMenu::LiquidMenu(LiquidCrystal_I2C&)'`
how can I fix it? library was downloaded from IDE

VaSe7u, 2020-01-11:
Because this is the I2C example you need to switch the configuration to I2C. In `LiquidMenu_config.h` comment lines 33-35 and uncomment lines 43-45.

closed 2020-01-11 by VaSe7u

antwal, 2020-01-12:
Not working:

```
/*!
 * @name Arduino's parallel "LiquidCrystal" library
 * @{
 */
// #define LIQUIDMENU_LIBRARY LiquidCrystal_LIBRARY
// #include <LiquidCrystal.h>
// #define DisplayClass LiquidCrystal
//!@}

/*!
 * @name I2C library
 * @see https://github.com/johnrickman/LiquidCrystal_I2C
 * @{
 */
#define LIQUIDMENU_LIBRARY LiquidCrystal_I2C_LIBRARY
#include <LiquidCrystal_I2C.h>
#define DisplayClass LiquidCrystal_I2C
//!@}
```

Error:

```
In file included from /Users/antwal/Documents/Arduino/libraries/LiquidMenu/src/LiquidLine.cpp:30:0:
/Users/antwal/Documents/Arduino/libraries/LiquidMenu/src/LiquidMenu.h:56:122: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal_I2C' (I2C) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal_I2C' (I2C) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                          ^
/var/folders/t8/b6xtsj5121q_9qcnbwcpcxwc0000gn/T//ccbVoEN0.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_sketch_jan12a.ino.cpp.o.2759':
<artificial>:(.text.startup+0x142): undefined reference to `LiquidCrystal_I2C::LiquidCrystal_I2C(unsigned char, unsigned char, unsigned char)'
/var/folders/t8/b6xtsj5121q_9qcnbwcpcxwc0000gn/T//ccbVoEN0.ltrans0.ltrans.o: In function `setup':
/Users/antwal/Documents/Arduino/sketch_jan12a/sketch_jan12a.ino:76: undefined reference to `LiquidCrystal_I2C::init()'
collect2: error: ld returned 1 exit status
```

andrwgldmn, 2020-01-12:
@antwal no, it`s working correctly with that editions.

VaSe7u, 2020-01-12:
Have you downloaded the [LiquidCrystal_I2C library](https://github.com/johnrickman/LiquidCrystal_I2C).

antwal, 2020-01-12:
yes use same lib but get this error always

VaSe7u, 2020-01-12:
The error states that there is an undefined reference to a function from this library.

Have you tried running an example from the I2C library? Maybe it's not installed correctly.
Do you get an error when I2C is disabled in LiquidMenu examples?

antwal, 2020-01-12:
not working on platformio, but working in arduino ide

antwal, 2020-01-12:
fixed