VasilKalchev/LiquidMenu #42: delete_Variable() and detach_Function()

gvisca44, 2020-08-24:
Forgive me ... first post on Github.  Nice library ... 

Given we have the facility to add a variable - might it be nice to delete too ?

Also, given we can attach a function - ability to detach might be nice ?

Cheers, 

Glenn.

VaSe7u, 2021-10-31:
Hello Glenn.

What use case do you have in mind for this functionality?

I'm hesitant to implement these methods without a use case, since a `remove()` method will have to rearrange the array in which the added variables are stored.

closed 2021-10-31 by VaSe7u

gvisca44, 2021-11-04:
Oh !  almost 12 months since I posted this.  

I am attempting to implement with a pretty deep menu structure - to display and update many values that ultimately get stored in EEPROM (measured in 10s or 100s of stored parameters).  

At the time I was creating the prototype code I think I was attempting to avoid code bloat by adding/removing or attaching/detaching different variables/functions based on the parameter being updated (rather than creating a new object for each one).

Glenn.