VasilKalchev/LiquidMenu #46: Switch focus on screens last line switches on not existing position

feoktant, 2021-07-01:
**Describe the bug**
Screen having 4 lines on last line `switch_focus` switches to position 4. On screen focus arrow just disappears. Then, on position 0 it switches back to the first line.

**Related code**
```
#include <LiquidCrystal_I2C.h>
#define LIQUIDMENU_LIBRARY LiquidCrystal_I2C_LIBRARY
#define DisplayClass LiquidCrystal_I2C
#include <LiquidMenu.h>

#define FUNC_NUM 1

LiquidCrystal_I2C lcd(0x27, 20, 4);
LiquidMenu menu(lcd);

LiquidLine line1(1, 0, "Setting 1");
LiquidLine line2(1, 1, "Setting 2");
LiquidLine line3(1, 2, "Setting 3");
LiquidLine line4(1, 3, "Setting 4");
LiquidScreen settings_screen(
  line1, 
  line2, 
  line3, 
  line4);

void setup() {
  Serial.begin(9600);

  lcd.init();
  lcd.backlight();

  settings_screen.set_focusPosition(Position::LEFT);
  line1.attach_function(FUNC_NUM, doNothing);
  line2.attach_function(FUNC_NUM, doNothing);
  line3.attach_function(FUNC_NUM, doNothing);
  line4.attach_function(FUNC_NUM, doNothing);

  menu.init();
  menu.set_focusPosition(Position::LEFT);
  menu.add_screen(settings_screen);
  menu.update();
}

void doNothing() {}

void loop() {
  menu.switch_focus(); 
  Serial.println(menu.get_focusedLine());
  delay(3000);
}

```
[image]

**Version of the library**
1.5.1

feoktant, 2021-07-04:
Such change will make it work
```cpp
// in LiquidScreem.cpp::switch_focus commenting out:
		if (forward) {
			if (_focus < _lineCount) {
				_focus++;
				// if (_focus == _lineCount) {
				// 	break;
				// }
			} else {
				_focus = 0;
			}
		} else { //else (forward)
			if (_focus == 0) {
			// 	_focus = _lineCount;
			// 	break;
			// } else {
				_focus--;
			}
		} //else (forward)
```
https://github.com/VaSe7u/LiquidMenu/blob/master/src/LiquidScreen.cpp#L138-L154

DotBreno, 2021-07-11:
Hi, @feoktant, thanks for sharing your solution. Your fix works for me when scrolling backwards, but the "ghost position" still exists when scrolling forward. If I scroll till the end of the screen and keep scrolling, the cursor disappears before appearing back on the first line. Am I missing something? Your changes fixed that for you? 

Thanks one more time for the effort

lbaraldi, 2021-09-20:
I don't know if it will cause any problems in another part of the library!

But this change worked fine for me....

https://github.com/VaSe7u/LiquidMenu/blob/master/src/LiquidScreen.cpp#L135-L157

```
void LiquidScreen::switch_focus(bool forward) {
	print_me(reinterpret_cast<uintptr_t>(this));
	do {
		if (forward) {
			if (_focus < _lineCount) {
				_focus++;
				if (_focus == _lineCount) { 
				 _focus = 0;
				//break;
				}
			} else {
				_focus = 0;
			}
		} else { //else (forward)
			if (_focus == 0) {
				_focus = _lineCount - 1; 
				//break;
			} else {
				_focus--;
			}
		} //else (forward)
	} while (_p_liquidLine[_focus]->_focusable == false);
	DEBUG(F("Focus switched to ")); DEBUGLN(_focus);
}
```

VaSe7u, 2021-10-31:
Duplicate with #38. Resolved with #39.

closed 2021-10-31 by VaSe7u