VasilKalchev/LiquidMenu #86: LiquidCrystal not found even if it's installed

Mark-81, 2022-03-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues.

### Library version

Yes, from Arduino library manager.

### Bug description

I create a new project using your example HelloWorld.
It does not compile because it complains about the missing of LiquidCrystal library.

As you can see from the output below the LiquidCrystal library is installed, detected and included. Still the compiler does not find it from LiquidMenu.h.

I'm running Visual Studio Code 1.65.2, PlatformIO 5.2.5 under Ubuntu 21.10.

### Compiler output

```shell
> Executing task in folder USBMidi: platformio run --environment leonardo <

Processing leonardo (platform: atmelavr; board: leonardo; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/leonardo.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Leonardo
HARDWARE: ATMEGA32U4 16MHz, 2.50KB RAM, 28KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.0.7
|-- <LiquidMenu> 1.6.0
|-- <MIDIUSB> 1.0.5
|-- <EncButton> 1.21.0
Building in release mode
Compiling .pio/build/leonardo/src/main.cpp.o
Compiling .pio/build/leonardo/lib675/LiquidMenu/LiquidLine.cpp.o
Compiling .pio/build/leonardo/lib675/LiquidMenu/LiquidMenu.cpp.o
Compiling .pio/build/leonardo/lib675/LiquidMenu/LiquidScreen.cpp.o
In file included from .pio/libdeps/leonardo/LiquidMenu/src/LiquidLine.cpp:30:0:
.pio/libdeps/leonardo/LiquidMenu/src/LiquidMenu.h:53:10: fatal error: LiquidCrystal.h: No such file or directory

***********************************************************************
* Looking for LiquidCrystal.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LiquidCrystal.h"
* Web  > https://registry.platformio.org/search?q=header:LiquidCrystal.h
*
***********************************************************************

 #include <LiquidCrystal.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/main.cpp:5:0:
.pio/libdeps/leonardo/LiquidMenu/src/LiquidMenu.h:52:123: note: #pragma message: LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.
 # pragma message ("LiquidMenu: Selected 'LiquidCrystal' (parallel) library. Edit 'LiquidMenu_config.h' file to change it.")
                                                                                                                           ^
*** [.pio/build/leonardo/lib675/LiquidMenu/LiquidLine.cpp.o] Error 1
In file included from .pio/libdeps/leonardo/LiquidMenu/src/LiquidScreen.cpp:30:0:
.pio/libdeps/leonardo/LiquidMenu/src/LiquidMenu.h:53:10: fatal error: LiquidCrystal.h: No such file or directory

***********************************************************************
* Looking for LiquidCrystal.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LiquidCrystal.h"
* Web  > https://registry.platformio.org/search?q=header:LiquidCrystal.h
*
***********************************************************************

 #include <LiquidCrystal.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/leonardo/lib675/LiquidMenu/LiquidScreen.cpp.o] Error 1
In file included from .pio/libdeps/leonardo/LiquidMenu/src/LiquidMenu.cpp:30:0:
.pio/libdeps/leonardo/LiquidMenu/src/LiquidMenu.h:53:10: fatal error: LiquidCrystal.h: No such file or directory

***********************************************************************
* Looking for LiquidCrystal.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LiquidCrystal.h"
* Web  > https://registry.platformio.org/search?q=header:LiquidCrystal.h
*
***********************************************************************

 #include <LiquidCrystal.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/leonardo/lib675/LiquidMenu/LiquidMenu.cpp.o] Error 1
======================================================= [FAILED] Took 0.43 seconds =======================================================
The terminal process "platformio 'run', '--environment', 'leonardo'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

Futski-III, 2022-05-19:
I'm wondering if there has been any resolution for this.  I'm having the same issue except with LiquidCrystal_I2C.h

### Library version: 
Yes, from PlatformIO library manager. -vase7u/LiquidMenu@^1.6.0

### Bug description:
Compile returns message stating that it can't find #include <LiquidCrystal_I2C.h>.
I am using https://github.com/johnrickman/LiquidCrystal_I2C library as specified in the LiquidMenu_config.h file.
In addition, Platformio keeps reinstalling the Arduino standard LiquidCrystal.h library even though all references to it have been removed.  I'm not sure if there is a hidden dependency to that library somewhere.  I have noticed that the Arduino standard also has a LiquidCrystal_I2C.h.   

If I remove LiquidMenu, and do a simple LiquidCrystal_I2C write, the program will complie and work successfully with the existing LiquidCrystal_I2C library.  This Ex is successful,

// This Example works successfully,
```
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd 0x27, 20, 4);
void setup() {
 lcd.init();                      // initialize the lcd 
 lcd.backlight();
 lcd.setCursor(3,0);
 lcd.begin(20, 4); 
  lcd.print("Hello, world!");
}
```
// But this Example fails,
```
#include <LiquidCrystal_I2C.h>
#include <LiquidMenu.h>
LiquidCrystal_I2C lcd 0x27, 20, 4);
void setup() {
 lcd.init();                      // initialize the lcd 
 lcd.backlight();
 lcd.setCursor(3,0);
 lcd.begin(20, 4); 
  lcd.print("Hello, world!");
}
```

### Platformio.ini as follows
```
[env:pro16MHzatmega328]
platform = atmelavr
board = pro16MHzatmega328
framework = arduino
monitor_speed = 115200
lib_deps = 
	paulstoffregen/Time@^1.6.1
	marcoschwartz/LiquidCrystal_I2C@^1.1.4
	vase7u/LiquidMenu@^1.6.0
```
### I have adjusted LiquidMenu_config.h as follows:
```
//==========================
// #ifndef LIQUIDMENU_LIBRARY
//     /// Wrapped library ID
//     #define LIQUIDMENU_LIBRARY LiquidCrystal_LIBRARY
// #endif
// #ifndef DisplayClass
//     /// Name of wrapped library's class
//     #define DisplayClass LiquidCrystal
// #endif
//!@}

/*!
 * @name I2C library
 * @see https://github.com/johnrickman/LiquidCrystal_I2C
 * @{
 */
#ifndef LIQUIDMENU_LIBRARY
    #define LIQUIDMENU_LIBRARY LiquidCrystal_I2C_LIBRARY
[... 87 lines cut ...]
* Web  > https://registry.platformio.org/search?q=header:LiquidCrystal_I2C.h
*
***************************************************************************

 #include <LiquidCrystal_I2C.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pro16MHzatmega328\lib9e1\LiquidMenu\LiquidMenu.cpp.o] Error 1
In file included from .pio\libdeps\pro16MHzatmega328\LiquidMenu\src\LiquidSystem.cpp:30:0:
.pio\libdeps\pro16MHzatmega328\LiquidMenu\src\LiquidMenu.h:56:10: fatal error: LiquidCrystal_I2C.h: No such file or directory

***************************************************************************
* Looking for LiquidCrystal_I2C.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LiquidCrystal_I2C.h"
* Web  > https://registry.platformio.org/search?q=header:LiquidCrystal_I2C.h
*
***************************************************************************

 #include <LiquidCrystal_I2C.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pro16MHzatmega328\lib9e1\LiquidMenu\LiquidSystem.cpp.o] Error 1
===================== [FAILED] Took 3.93 seconds =====================
```

burak1000, 2022-08-01:
I found the solution

:

in
LiquidMenuconfigfile   uncomment other library. and like this

/*!
 * @name Some other library
 * @{
 */
 #include "FUCK_I2C.h"
 #ifndef DisplayClass
 #define DisplayClass FUCK_I2C
 #endif
//!@}
//!@}

after that copy original LiquidCrystal_I2C.h and LiquidCrystal_I2C.cpp file and replace all  LiquidCrystal_I2C words to FUCK_I2C
also change all file name like this FUCK alternate.  also folder name too

and delete library specific files. just leave h. and cpp file.

and add this folder your lib directory of your project.

this will force platform io use this library. after 4 hours

closed 2022-12-14 by VaSe7u